glazet/QuadEVCharger #5: Who are Potential Customers

neilh10, 2023-09-17:
Issue can be used as tracking items. I'm generating a few here to be tracked and discussed

Neil:  Who are the potential customers - the people who specify it and/or buy it.

From Bab Solater's email 2023-09-16, 
4.	Develop some understanding as to who may be potential customers and
the business models that are likely to be deployed, e.g.:

a.	End customers: Property owners/developers/property managers
b.	“Intermediate customers” eg. Design/Build Contractors

CalvinJDP, 2023-09-17:
Our potential customers would be property developers/contractors, property owners, businesses with parking facilities, and EV fleet owners.

glazet, 2023-09-17:
Also EV charger manufacturers could want a quad charger. I have only found dual chargers being sold. I'm not sure if EV manufacturers would be interested in a charger that isn't their proprietary model. After reading an NPR article on EV charger demand, city municipalities could want these chargers to install on the streets.

neilh10, 2023-09-18:
@CalvinJDP  good, needs maybe more flushing out.  At the core - potential customer are those who have 4 parking spaces that a consumer vehicle might plug into and be left for 4+ hours.  So multi-family overnight, workplace parking lots for daytime parking, Ferry Terminal parking, commuter park and ride, Then its the process who designs for these spaces and who owns them.

neilh10, 2023-09-18:
> Also EV charger manufacturers could want a quad charger. I have only found dual chargers being sold. I'm not sure if EV manufacturers would be interested in a charger that isn't their proprietary [...]

@glazet  I think should track all the EV chargers you find https://github.com/glazet/QuadEVCharger/issues/9

closed 2023-11-07 by glazet