glazet/QuadEVCharger #8: What functionality does the common control need to support

neilh10, 2023-09-18:
This is a tracking item to collect the internal engineering technical requirements for the common control.
This may map to items specified in https://github.com/glazet/QuadEVCharger/issues/7 

For instance each charging port has technical specifications to detect a charging cable plugged in, and also through a 12V 1KHz PWM. Provide a technical reference for each specification  

This is likely to feed into the decision on what small board computer to use, and inputs and outputs it needs to support, with custom hardware needed

glazet, 2023-09-21:
I have been looking into what microcontrollers we can use to output 4 PWM signals.  I have found a few:

ESP32 - Has 16 channels for configurable PWM signals. See page 36. ESP has a large community so resources should be manageable to find. Currently using ESP8266 in my IoT class, will soon develop comfortability with this chip.
chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

Arduino Mega 2560 Rev3 - This has 15 pins that can be used as a PWM channel. Its Arduino so heavy support as well.
chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://docs.arduino.cc/resources/datasheets/A000067-datasheet.pdf

PIC18F46K42 - A PIC microcontroller that Django and I are familiar with from our microcontrollers class. It has 4 PWM channels which is exactly what we need. I have 6 of these on me so they'll be cheap(free) and replaceable. See page 11 for PWM pin data. 
chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/http://ww1.microchip.com/downloads/en/DeviceDoc/Microchip%208bit%20mcu%20%20PIC18%20L_F26_27_45_46_47_55_56_57K42%20low-power%20high-performance%20mcu%20with%20xlp%20tech%2040001919B.pdf

RaspberyPi 3B+ or higher models - we could use the GPIO pins to output a PWM signal but takes some coding finesse. Could be useful for network connection. Easily access WiFi. May not be perfect for relay control.

djangodemetri, 2023-09-21:
[Manual with tolerances (PDF page 41)](https://hmr.araiindia.com/Control/AIS/210201752832PMAIS138_Part1.pdf)

Requirements:
1) 4 pins capable of generating PWM signals
2) 4 ADC pins with high enough resolution (maybe 10-bit minimum) and fast enough processing time (shouldn't be a problem in our application)
3) 4 pins for Proximity Pilot
4) 4 pins for triggering relays/contactor
5) Not necessary but for future proofing, should be compatible with either 
     a) CAN bus
     b) Wi-Fi
     c) Bluetooth
6) Likely need watchdog capabilities 
7) Power saving would be nice when no EVs are plugged in

neilh10, 2023-09-21:
> I have been looking into what microcontrollers we can use to output 4 PWM signals. I have found a few:

JUST an FYI, when looking at the processor, I would suggest you look at it as a SBC - Single Board Computer - integrated computer and circuitry with a connector.  
List all the specs needed first in table format with a weighting. So program size, 4 PWM capable (may be seperate board, but be nice if onboard, likely to requires specialist interface circuitry), Connectivity/Wifi, USB ports (possible Screen interface),  should it support lunix, power used and of course cost. 

There are many SBC lists out there, and once you learn a specific https://www.seeedstudio.com/blog/2019/11/20/best-single-board-computers-of-2019/
https://www.electromaker.io/blog/article/best-single-board-computers

Then you are looking for the 4 PWM support,   and then software that would support testing the hardware.
So just as navigator there is Raspberry Pi flavors, there is also Beagle Bone Green Wireless (that I like and have used)  
There are some good reasons to go with a non-linux machine, like needing to be power conscious, but this doesn't apply with your power capability.
So you may be looking at types of linux to consider, from Raspberian, BeagleBone, Ubuntu
https://ubuntu.com/embedded
So my observations is this will knock out the small microcontrollers MEGA2560 PIC18F46K42 
I would think the ESP32-S3 would make your list ~ and a SBC list . https://shop.m5stack.com/collections/m5-controllers
https://www.cnx-software.com/2021/07/18/linux-5-0-esp32-processor/

neilh10, 2023-11-06:
https://github.com/glazet/QuadEVCharger/discussions/18#discussioncomment-7484395 mentions prototyping with ESP32-S3 PWM as being a significant factor. 
Seems like that maybe should be a separate **dependent** issue linked to this and #22.

Doing that you could  list the tools you are using and what the status is, ie not working

I've been looking closely at the ESP32-S3 and particularly its dual processor and co-processor RISC and the new variant ESP32-C6. The ESP32-S3 Looks to be nice choice for QuadEV, with allocation of the processors to specific current and future tasks.

There is a https://shop.m5stack.com/products/m5stack-cores3-esp32s3-lotdevelopment-kit with 2" screen that might fit with a QuadEV proto.

Earlier this year I purchased the M5Stack - Station https://shop.m5stack.com/products/m5stack-station-esp32-iot-development-kit-rs485-version for a trial which hasn't worked out. It has the dual core ESP32-D0WDQ6-V3, and I could loan that if its useful.

I've been using tools platformio.com  which is a productivity environment for Arduino Framework, and sits on top of VSC. 
However it needs tweaking before the  standard Arduino builds. Happy to help with the learning curve
https://docs.m5stack.com/en/quick_start/cores3/arduino

neilh10, 2023-11-07:
Good to see the PWM working test on ESP- S3
See [com/products/m5stack-cores3-esp32s3-lotdevelopment-kit](https://shop.m5stack.com/products/m5stack-cores3-esp32s3-lotdevelopment-kit)

test software 
https://github.com/m5stack/M5CoreS3

closed 2024-04-14 by glazet