glazet/QuadEVCharger #21: Proximity Pilot

glazet, 2023-10-23:
The first connection with the EV and EVSSE, this is a crucial step in our project. Here we can discuss how to build and implement the proximity pilot circuit and effectively utilize its status.

neilh10, 2023-10-23:
Can you describe the Proximity Pilot in terms of the specification. Circuit diagram etc can be posted. Any precursor (like ADC) can be listed.  The Labels - can be set to enhancement, the Assigneess to team member is working on it.

glazet, 2023-10-26:
[image]
^Info on PP voltage states

[image]
[image]
[image]
^Calculations for Proximity pilot voltages

neilh10, 2023-10-26:
Good to see the how the cable carrying capacity is defined. 
Its built into the J1772 Plug as part of its handle with the switch - when the switch on the top of the handle is pressed it tells the vehicle to stop taking a charge.
From what I've seen, the vehicle then has a mechanical latch, that unlocks the plug , and the plug can then be manually removed.
As per the email - Oct 24 - seems to me since the  Proximiy Pilot circuitry is located in the cable head "EVSE Connector" - its purchased in as part of the J1772 cable, and I don't believe your hardware verification goes to having a vehicle present  - it doesn't need to be designed. Magic - best type of circuitry - part of an item purchased in.