glazet/QuadEVCharger #22: Control Pilot hub

glazet, 2023-10-23:
Here we can discuss how we will construct our +/- 12V 1kHz square wave that is within +/-4Hz accuracy.  We must be able to have this signal reliably reach each of the 4 EVs.

neilh10, 2023-10-25:
To be accurate, each EV cable requires a CP generator, under the control of the processor
From https://en.wikipedia.org/wiki/SAE_J1772 - references SAE J1772-2017 standard , and a general circuit. There maybe a more authorative reference, but this is the general requirement 
[image]

glazet, 2023-10-26:
[image]
^Control Pilot states and their voltages

[image]
^Effect of duty cycle on amp level

[image]
^Example signal

Source: [https://www.fveaa.org/fb/J1772_386.pdf]

neilh10, 2023-11-06:
https://github.com/glazet/QuadEVCharger/discussions/18#discussioncomment-7484395 mentioned CP verification with simulation. I wonder could this be updated to reflect work done and data collected?

neilh10, 2023-11-13:
In https://github.com/glazet/QuadEVCharger/discussions/18#discussioncomment-7558644 issues where mentioned with the Op-amp  but no values are given. 
Please identify the op-amp you used and the test circuit in this issue report.

2nd request - please publish the circuit you are using for CP and simulation values. 
Using LTspice for the simulation is FANTASTIC !!!, It provide the opportunity to make real technical comments. 
  Its probably a good idea to maintain the same output impedance (sum of the resistors) with the shift to 3.3V.   
The resistors need to be taken from a series that is purchasable and have a defined tolerance - probably 0.1% in this case as its creating a measurable threshold, but maybe 1.0%,if you can't find a low cost 0.1%.  Probably E96 or E192  https://en.wikipedia.org/wiki/E_series_of_preferred_numbers

The effect of what happens when the vehicles charging circuit activates should also be included in the simulation. 
For a comprehensive simulation it would also be useful to simulate the tolerance on the voltage lines +12V and -12V and +3.3V.  These should be possible to fit into the simulation
https://courses.engr.illinois.edu/ece343/sp2021/tutorial/ltspice_tut.pdf  - possibly "Monte Carlo "

These are all needed to determine what are the DC ranges that the ADC needs to read, and is the ADC Vref stable and accurate enough for reading these ranges. 
https://github.com/glazet/QuadEVCharger/issues/19

neilh10, 2023-11-14:
Great to hear and that you are getting real world issues. Thats the puzzle.  
The ADA4610 is a "family" so you  need to order a specific device and you want it to meet outdoor temperature?  
https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4610-1_4610-2_4610-4.pdf
It looks to me there isn't a DIP version.
How are you planning to breadboard the specific device ordered? is it DIP or SOIC.  

quad OpAmps  are very plentiful - and need the higher voltage of +24V but low slew rate - simulations are all similar for this type of frequency
From Digikey I count 236  https://www.digikey.com/en/products/filter/instrumentation-op-amps-buffer-amps/687?s=N4IgjCBcoCwdIDGUBmBDANgZwKYBoQB7KAbRAGYBOAJhgFYAGEAXQIAcAXKEAZQ4CcAlgDsA5iAC%2BBMJQAclKKGSR02fEVIgmzKSGoNKMRUlSZcBYpDJHWITtwCSwjjlE5%2BkgtQBsYJgmVVcw0rPRgAAgA1EC9vKJiKBniCcjjoghgk9JAYCOjbe0gQAFVhQQ4AeRQAWRw0LABXfhxPEABaamNlAQb1SzI6FgldTtCKtjQAWzYsIaA
Though there are 19 that are a DIP package often useful in prototyping on a bread board, and even smaller number that are outdoor range.

Possibly the LT1491CN is an analog device (maybe has a simulator) and comes in 14PIN DIP for breadboarding - though still on the expensive side if this was a real product.
https://www.analog.com/media/en/technical-documentation/data-sheets/14901fb.pdf
[LT1491CN#PBF](https://www.digikey.com/en/products/detail/analog-devices-inc/LT1491CN-PBF/1620006)

For your problem there isn't enough detail  - " no voltage reading at ADC-Vref" - measured by what ? Voltmeter or internal ADC.   Be good to snip the circuit and post it. :)

neilh10, 2023-12-03:
How about this analysis of the CP circuit - could it be discussed at the next meeting
[QuadEV_CP_RRR_divider - 5V.xlsx](https://github.com/glazet/QuadEVCharger/files/13539770/QuadEV_CP_RRR_divider.-.5V.xlsx)

djangodemetri, 2024-02-23:
Part 1 is an initial look at the circuit. Part 2 is the simplification using the assumptions made on the right.

I'm not sure my handwriting is legible so I'm going to transcribe what I wrote here:
"But the comparator saturates and does not display a typical IV curve. So we can approximate it as a independent voltage source and internal resistance. Also Rin of the op-amp is almost infinitely large so the rest of the circuit is isolated.

[image]

neilh10, 2024-02-23:
[image]

So assuming you want to have the ADC swing between 3.3V and close to 0V  (though actually it maybe that the input range when calibrated is actually 2.8V?) 

The values you have posted of 140K 66K 40K  with 3.3V pullup for the +12V to 12V,   I get a swing of 3.521V to 0.018V   - so upper end out of range.

Now should be noted the 66K and 40K don't exist as real resistors - the E96 series lists 66.5K and and 40.2K

https://www.vishay.com/docs/28372/e-series.pdf

The standard values of 200K+100K - 300K total impedance you probably don't want to change - ie you have 140 + 66=206K has dropped the interface loading  by 1/3

For some values I tweaked - I got a swing

218K  81.6K 58.3K  for +12V to -12V I get a swing of 3.284V and 0.041V     ~ and impedance is 218+81.6 299.6K - close to the original 300K

[QuadEV_CP_RRR_divider.xlsx](https://github.com/glazet/QuadEVCharger/files/14390893/QuadEV_CP_RRR_divider.xlsx)

I could only find a device OPA4227 for quad opamp
Looking at the circuit 
https://www.fveaa.org/fb/J1772_386.pdf
the Diode is a clamping diode preventing a negative swing, not a rectifying in series diode.

neilh10, 2024-02-24:
The original reference circuit is Pg 36 on https://www.fveaa.org/fb/J1772_386.pdf
Correction : the diode shown is a clamping TVS diode, "22.5V Clamp 27.1A Ipp Tvs"
https://www.digikey.com/en/products/detail/littelfuse-inc/P6SMB16A/1766582
 probably to protect the Op-amp from any spikes on the PILOT  cable that goes out to the vehicle.

[image]

CalvinJDP, 2024-02-27:
Currently in our schematic we have a rectifier circuit between the S3 and the resistor network. If we use your suggested resistor values does that mean we do not need the rectifier circuit? 
[image]

neilh10, 2024-02-27:
yes. If you already have the PCB, it can just be shorted.
The interesting part of what the resistor network is doing -  level shifting it to be in range of the processors. 
The original circuit was 0-5V - and using the spreadsheet I created a thevinens equivalent circuit with the resistors - and it shifts between . Now its 0-3.3V
Initially when I saw it, I thought how is it coping with going negative.
Turns out with the original resistors ~ 
see QuadEV_CP_RRR_divider - 5V.xlsx
Change cell E25 for +12 and then -12 - and read off E56
+12V in resulted in +4.543V at PilotRead , and 
-12V resulted in  0.891V at PilotRead

Then change the resistor values and Vcc=3.3V in
QuadEV_CP_RRR_divider.xlsx 
I went over with @djangodemetri on Saturday.

neilh10, 2024-04-11:
Good to see the full schematic V3 on - https://github.com/glazet/QuadEVCharger/tree/main/Schematics
 One circuit "J1772 negotiation", could be good to highlight how you verification tested   
with all the electronic parts and values the team decided on is the path of the PWM.  Explaining the function of the 5 resistors - 4 on the board below, and one on the EV - R19 current limit, R17/R16 voltage divider, R18 bias resistor. 
Why PWM swings of up to +12V to -12V from OPA1 driven by the processor , are seen as 0-3.3V at the ADC,  and what values you calculate in the 0-3.3V range and what values you actually measured for circuit verification

[image]