glazet/QuadEVCharger #23: Decision Algorithm for EVSE

glazet, 2023-10-23:
Here we can discuss how the algorithm will decide who gets charging and when they receive it. Utilizing user inputs would allow us to make better educated decisions on who should get charging. Keeping in mind that the issue we are aiming to solve is long term parking, so those who need charge and will remain in their spot for 4+ hours.

neilh10, 2023-10-31:
I would suggest this be broken into a number of sections. a) controlling the hardware EVSE CHARGE PILOT - Advertising current available,  (ADC level input and PWM Duty Cyclable)   implemented as a "Firmware Driver"  b) the User Interface  c) the billing output?  d) ? 
etc. For 
The file https://github.com/glazet/QuadEVCharger/blob/main/Algorithm.py probably belongs to this section. 
For a) Can this be described in terms of  Driving the PWM https://github.com/glazet/QuadEVCharger/issues/22#issuecomment-1781831137 - Pseudo Code in the file could be used.
[Possibly https://www.fveaa.org/fb/J1772_386.pdf Pg13 expected PWM levels   Pg 36 circuit interfaces)

neilh10, 2023-11-03:
The OpenEVSE is quite a find. It implements a full algorithm on the AVR processor and the code is available for looking at https://github.com/OpenEVSE/open_evse/tree/master/firmware/open_evse. 
Notice the https://github.com/OpenEVSE/open_evse/tree/master  platoformio.ini - indicating it uses the platformio.org build environment. I've been very pleased with it myself.