glazet/QuadEVCharger #24: Switching Devices Hub

glazet, 2023-10-31:
Here we can discuss our switching device and its control parameters.

glazet, 2023-10-31:
Here is the current contender for our switching device. The [Packard C240C](https://store.openevse.com/collections/all-products/products/packard-c240c-40a-contactor-208-240vac-coil-50-60hz) is a 208/240VAC contactor with two poles. It requires 12VDC to activate the switch. It can handle 40A inductive and 50A resistive. Along with its cheap price per unit(~$11), it has been used by other EVSE projects and works wonderfully. Only concern is that I have not been able to find a datasheet but I have contacted Packard themselves and have requested one. Other relays I have found either don't have a 50A max capacity or are too expensive to justify buying($108 per unit).

glazet, 2023-11-14:
We have finally found and ordered our relay we will use. The STRUTHERS-DUNN 450XBX50-12VDC will be able to meet our design parameters of 240VAC 50A switching requirements and takes 12-VDC to open and close the poles. We only ordered one right now to be used for testing. Our initial tests will be opening and closing the poles using the ESP32. This will require a driving circuit, which is provided on the OpenEVSE PowerPoint. We have the correct transitors needed for the driving circuit so its just a matter of when things get delivered. I can begin with a simulation of the driver circuit on LTspice. 

[image]
^Image of the relay driver circuit diagram

glazet, 2023-11-14:
My question for @neilh10 is what is the JB1 component in this diagram. My assumption is that is the relay itself but as a SPST version, but I'm not 100% sure. I have not seen that component symbol before.

neilh10, 2023-11-14:
@glazet - did you mean  JP1 - probably the socket for the relay of the original circuit. Its a typical connector symbol - 3pin. May be different for the 450XBX50-12VDC
My guess for the picture/cct shown its the relay driver. The relays look like they have a spade connection for the 50A path.

https://relays.struthers-dunn.com/item/power-relays/450-series-cross-reference/450xbx50-12vdc
Seems like for safety you would want the Cover or  450XBX50C-12VDC
[image]
You might want to have the LED that indicates its on. Do you have an application not for the relay - I can't see how the wires connect to it, seems like it is a socket.
These heavy duty relays are challenging, though there must be a version that is in all the different EV chargers out there