glazet/QuadEVCharger #26: Plausibility of Engineering Requirements

djangodemetri, 2023-11-13:
ER 3: Must be capable of switching current nominally at 208VAC
1. We need door access to the room with plug
2. Find wire that can handle 208VAC (not hard)
3. Double check that we have equipment that can read 208VAC. If not, we need to step the voltage down

ER4: Must be capable of switching current nominally at 25A
1. Find a device that can measure 25A. If one is not available, maybe create a current divider and measure each branch separately (4 equal resistors, each having 6.25A?)
2. Calculate how much power each resistor would have and buy resistors that would rated at that power. (6.25A*5V=31.25W?)
3. Make sure the power supplies can actually reach 25A. Otherwise, change the ER to the maximum amperage the power supply can handle

ER7: Must be able to visually signal an alarm state that is visible within 10ft, line of sight.
1. How do we actually test this? 

ER8: Must satisfy IEEE 802.11b/g/n
1. Research exactly what is required to satisfy these requirements. 
2. Do we have the equipment needed to measure this?

ER9: NFC device must satisfy ISO 10373-6 (or can read 95%)
1. Likely we will not be able to find ISO 10373-6 so, in reference to Neil's troubles with his card not scanning, we could perform a reliability test. Can it successfully read data 95% over 50 scans?

neilh10, 2023-11-13:
TO me this is WAY too many technical issues for ONE tracking item. 
The purpose of a tracking item is to be able to IDENTIFY a problem you need to track,  come up with SOLUTIONS and then IMPLMENT them.   
The life cycle of the issue, is to initiate the item, allocate an owner at various stages, for that owner to find solutions, and or pass to another person in the team, before a team member implements the SOLUTION and closes it.

I would recommend you  create separate items for each of these technical issues, and then allocate owners, OR assign to yourself.

djangodemetri, 2023-11-14:
An issue for each question or an issue for every engineering requirement, (consisting of multiple questions)?

neilh10, 2024-04-11:
Is this still useful for the project?

closed 2024-04-14 by glazet