glazet/QuadEVCharger #28: Current Sensing

djangodemetri, 2024-02-16:
This will consist of two parts: first, what I have done and second the questions I have.

Part 1: Progress?

Here is the circuit for the current transformer:
[image]

R1 is a burden resistor, which loads the CT and provides it a specific accuracy, per the datasheet. D1 and C1 make a simple passive rectifier. R2 and R3 will be a voltage divider because the CT outputs 5V when reading 50A. The values for R2 and R3 are yet to be found because a potentiometer was being used. (Yes, this may also change the value of C1 when R2 and R3 are picked)

The burden resistor was calculated using the equation, (which was found [here](https://electronics.stackexchange.com/questions/124386/calculating-a-burden-resistor-for-a-current-transformer#:~:text=Alternatively%2C%20if%20you%20really%20want,the%20burden%20resistance%2C%20in%20ohms.)):
[image]
I realize now that the [datasheet](https://amgistoroids.com/wp-content/uploads/2020/03/CT1050.pdf) has values:
[image]
The intersection for approximately 5V and 10A through 50A seems to given a burden resistor value of around 100 ohm to 200 ohm. Whoops.

The voltage divider was set up as a potentiometer. I decided I would design around the maximum value. I generated 5V dc at the burden resistor, (since 5V is the maximum value the current transformer will produce), and adjusted the potentiometer until voltage across R3 was 3.3V. 

Now that the circuit was built, it was time to test it. Sadly, the only ac power source that is available has a limit of 10A. I assumed the relationship between the CT voltage and the ADC voltage would be linear, so if I could plot points between 0A and 10A, I could extrapolate the entire relationship between ADC voltage and CT amperage.
[image]
This would mean that for every 1V at the CT, the CT would be reading 10A. If I corresponded each experimental ADC voltage value to a theoretical A value, I could obtain a constant and simply scale the reading the in the code. (This by no means is the most accurate but it's a starting point). But as you can see, it is not a linear relationship. 

Part 2: Questions

I have at least two tolerances that I need to figure out. As seen the [datasheet](https://amgistoroids.com/wp-content/uploads/2020/03/CT1050.pdf), the lower the burden resistor, (or I assume it's the burden resistor), the less percent error we have. Secondly, we have the error in the ADC. The ESP32 has a 12 bit ADC which gives us a resolution of .8mV [step = (3.3-0)/2^12]. These tolerances stack so do we just add them? 

How do I find the constant that I will multiply the ADC value by to convert it to current? How will I find this if the power supply can only achieve 10A and this relationship is not linear, (so it cannot be extrapolated)?

neilh10, 2024-02-16:
Hi Django   nice write up.  Really fantastic to see the circuit explained and look for a low cost way of measuring current

Great to see the calculations fleshed out - really need to do that for this type of circuit.

Seems to me there are four areas of error that still need to be investigated - the joy / challenge of analog circuitry :)   this is often why analog circuitry takes a lot of time to fine tune.  Sometimes can be monticarlo simulated for ranges of errors, but not all errors are built into simulators. 

1) ADC Vref error 2) Load Resistor accuracy error 3) load resistor Temperature Coefficient error 4) Diode + R1/R2  accuracy

4) Looking at his first - R1/R2 could be eliminated by using the internal attenuator/divider - see ESP32-S3 . 
Possibly provide a 3V3 clamp & R/C . 
The C1 is better being close to the ESP32-S3 ADC pin, and a series R to the load resiator can then protect some spike protection with C1 and not affect the accuracy of measurement.   
D1 is "text book", but a real problem for an accurate ADC - as the turn ON is likely to be 0.6V, and that means it can't measure the last 0.6V of voltage from the loading resistor. Lots of discussion about this - and possibly use germanium diode -the problem is having a calibrated turn-on - and its oscilating at 60Hz.
Another option might be to run the ADC faster than 60Hz, eg at 6Khz and look for the peak measure voltage over 8.3mS  1/(60*2) =. Then rely on the same  clamping diode to prevent the ADC_INPUT going negative and clamp spikes on loading resistor going above 3.3V. 

These sources of error affect large volume production - so they may not impact single unit prototypes, except for the first error- but are good to understand. 

1) The ADC Vref Error - From the ESP32S3 Technical manual 39.3.5 
Vdata =  Vref/4095  × Adc_Value 
https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pd
elsewhere - it references the Vref as nominally 1.1V ~ though need to find a reference for its accuracy
Some clues https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32s3/api-reference/peripherals/adc.html
I found "  however the true reference voltage can range from 1000 mV to 1200 mV"
that is the Vref is nominally 1.1V with a +/ accuracy of 0.1V  ~ or about +/-9%
and it suggests a way of calibrating the ADC - that is determine what the actual value is on this specific chip,
This would need to be calibrated per chip probably with an external meter of known accuracy
And that becomes the Vref_error.  If the external meter was +/1mV then the Vref_error=+/-1mV
Then that would need to be used in the above "Vref" 
It seems the ADC subsystem can then attenuate that input voltage to allow stepped voltages.
See https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32s3/api-reference/peripherals/adc.html#adc-attenuation
However if the actual Vref=1.0V then the highest voltage measurable would be 2.818V = 1.0/1.1 * 3.1V 

So for the loading circuit, you don't want an input voltage to go above 2.8V as otherwise for some ESP32-S3 it might become non-linear.
From the CT1050 loading graphs above eyeballing it seems like 40A would produce 2.0V and 50A closer to 2.5V - so a good item to calculate. 

2)  The loading resistor has an error - manufactured to a tolerance - its probably easy to go from 0.1% to 0.05% and possibly to  0.01% resistor and low thermal drift. Its low wattage as the max voltage on it is only likely to be 2.5V

3) Since this outdoor temperature range the load resistor - and other components will have temperature errors - or manufacturer characterizes it as a maximum "Temperature Coefficient". So a low temperature drift resistor is Metal Foil and Thin Film. Metal Foil is way more accurate and costly - probably used in space or aero space applications.
That is they have a value at -10C and then resistance changes as  the temperature increases to +70C.
For an accurate 0.01% at 49.9ohms at [Digikey](https://www.digikey.com/en/products/filter/chip-resistor-surface-mount/52?s=N4IgjCBcoMxaBjKAzAhgGwM4FMA0IB7KAbRADYBWCAXXwAcAXKEAZQYCcBLAOwHMQAvvjABOABwj4IJJDRY8hEiAAMIakJAAmZWIpSZcnPiKRSAFhEA6EQAIA8gAsAtphD4Ky%2B89fvllil4uavRMkCAAqtycDHbIALLYqJgAruzYgvgAtJr6UBzJCiakeuoaOaYgaZicmAwE7MHkUpwAJsyZYMoQIcxuIAwAnnTpYUlIAgJAA) (as they are the lowest cost ) the +/- 5ppm/C is $1.91
for a +/- 2ppm/C the cost doubles to $2.80 and there is even a lower+/ 0.2ppm - check out its cost.

Also started here https://github.com/glazet/QuadEVCharger/issues/20

neilh10, 2024-02-16:
analodRead() maybe pre-calibrated according to - https://esp32.com/viewtopic.php?t=34706, then just need to figure out what the error is.