glazet/QuadEVCharger #29: Porting ILI9341 display from ESP32-WROOM-32D to ESP32-S3

glazet, 2024-02-23:
There are some roadblocks to porting the current GUI setup from the EP32-WROOM-32D to the S3. The GUI was initially designed on the WROOM32D because of an abundance of support online in connecting it to the ILI9341 display. Documentation on S3 to ILI9341 is sparse and not very detailed.

glazet, 2024-02-23:
I am currently having issues even getting a simple demo to appear on the display. When I built the GUI on the WROOM32D I had to only uncomment lines 212-217 in the User_Setup.h file in the TFT_eSPI arduino library and connect the display to the corresponding pins. 
[image]

I have tried to change the pin numbers in the User_Setup.h to match those I desired on the S3. They were as follows:
CS     GPIO3
RST   GPIO46
DC    GPIO9
SDI   GPIO11
SCK  GPIO12

[image]

I have also tried assigning the pins in the main program code as shown below but this too did not get any results.
#include <SPI.h>

// Define GPIO pin numbers
#define MOSI_PIN 11  // Replace with your desired MOSI GPIO pin
#define SCLK_PIN 12  // Replace with your desired SCLK GPIO pin
#define CS_PIN   3  // Replace with your desired CS GPIO pin

void setup() {
  // Start SPI communication
  SPI.begin();

  // Configure GPIO pins for SPI
  pinMode(MOSI_PIN, OUTPUT);
  pinMode(SCLK_PIN, OUTPUT);
  pinMode(CS_PIN, OUTPUT);

  // Set initial state for CS pin 
  digitalWrite(CS_PIN, HIGH);  // Make sure the device is initially not selected
} 

Anyone have any ideas on what I could be missing here? Any help is appreciated.

glazet, 2024-02-24:
[Here](https://youtu.be/pYp1Nxmfdeg?si=VpHWhF6YtDlEoi5G) is the link to the resource used to setup the ILI9341 with the WROOM32D initially.
[Here](https://youtu.be/6FsFwa304qc?si=bUcBk8F3_7bl08zt) is a video I found that could potentially help with the S3 setup, although I've ran out of time today to try it out. Will try this when I'm free Sunday(2/25/24).

neilh10, 2024-02-24:
@glazet - gosh challenging it just doesn't work. Though that often happens.
What I do is try and find a basic example of something that does work, and then do little steps.
I would guess at it being the base S3 SPI driver setup - either in your wiring or in the low level software.
See if you can verify a simple program to get some status from ILI9341 ~ little steps to start with .

neilh10, 2024-02-25:
@glazet taking a quick peak at this ~ I guess I'm wondering about your expectation on mapping.

Shouldn't it also be declaring a MISO? 

1) Where did you get the board from, 
2) what tools are you using?
For instance I use PlatformIO/VSC  and with the project wizard I can select a new board "Espressif ESP32-S3-DevkitC-1-N8(8 MB QD, no PSRAM)
and Arduino Framework
And then it initializes and brings in the setup.

C:\Users\<user>\.platformio\platforms\espressif32\boards\esp32-s3-devkitc-1.json

and in platformio.ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

glazet, 2024-02-26:
@neilh10 - I have been taking the approach you suggested, the TFT_eSPI library has plenty of examples I've used to test the screen with. Sadly, I have been unsuccessful in getting any of these examples to work on the S3 but they easily run on the WROOM32. 

As for the not declaring MISO, I had never declared it on the WROOM32 and was able to get the display to work perfectly, as you've seen in person. I believe that the MISO would be used if we were using the touchscreen feature but we are not implementing that function.

I am using the Arduino IDE to code the S3, as I was not able to find board support for my ESP32-S3-DevKitC1-N16R8 on PlatformIO. I have made sure that the S3 is even getting programs loaded onto it by running Calvin's WIFI test and I do get results back. So I know the board manager works and the S3 can run programs. I believe my issue is in the header file for the TFT_eSPI library. I will take a deeper look tonight and see what I can do to get this working.

glazet, 2024-02-26:
I have successfully gotten a demo to work on the display with the S3. Human error was at fault. I will start rewiring the prototype to work with the S3.

neilh10, 2024-02-26:
@glazet hurrah!.  I do think you want to capture what works with a github repo.

closed 2024-02-26 by glazet